Electrotechnique/TSynth-Teensy4.1 #4: Typo in voice1On?

callimero, 2020-11-06:
In voic1One there is 

voiceMixer1.gain(0, VELOCITY[velocitySens][velocity] * level);

In all other voices it is:

voiceMixer1.gain(1, VELOCITY[velocitySens][velocity] * VOICEMIXERLEVEL);

`
void voice1On(byte note, byte velocity, float level) {
  keytracking1.amplitude(note * DIV127 * keytrackingAmount);
  voices[0].note = note;
  voices[0].timeOn = millis();
  **voiceMixer1.gain(0, VELOCITY[velocitySens][velocity] * level);**
  filterEnvelope1.noteOn();
  ampEnvelope1.noteOn();
  voices[0].voiceOn = 1;
  if (glideSpeed > 0 && note != prevNote) {
    glide1.amplitude((prevNote - note) * DIV24);   //Set glide to previous note frequency (limited to 1 octave max)
    glide1.amplitude(0, glideSpeed * GLIDEFACTOR); //Glide to current note
  }
  if (unison == 0)prevNote = note;
}

void voice2On(byte note, byte velocity, float level) {
  keytracking2.amplitude(note * DIV127 * keytrackingAmount);
  voices[1].note = note;
  voices[1].timeOn = millis();
  **voiceMixer1.gain(1, VELOCITY[velocitySens][velocity] * VOICEMIXERLEVEL);**
  filterEnvelope2.noteOn();
  ampEnvelope2.noteOn();
  voices[1].voiceOn = 1;
  if (glideSpeed > 0 && note != prevNote) {
    glide2.amplitude((prevNote - note) * DIV24);   //Set glide to previous note frequency (limited to 1 octave max)
    glide2.amplitude(0, glideSpeed * GLIDEFACTOR); //Glide to current note
  }
  if (unison == 0)prevNote = note;
}

`

ElectroTechnique, 2020-11-14:
Untrue, each voice has it's own mixer input on the the three mixers.

callimero, 2020-11-14:
Maybe I did not communicate very good:

On Voice one it is 

voiceMixer1.gain(0, VELOCITY[velocitySens][velocity] * **level**);  

On **all** other voices it is

voiceMixer1.gain(1, VELOCITY[velocitySens][velocity] * **VOICEMIXERLEVEL**);

So Voice 1 uses the level parameter, all others using the Constant VOICEMIXERLEVEL

Thats something I don't get.

ElectroTechnique, 2020-11-15:
Fixed in V2.02

closed 2020-11-15 by ElectroTechnique