Electrotechnique/TSynth-Teensy4.1 #5: Error compiling for board Teensy 4.1

dwestbury, 2020-11-12:
Can't seem to get the Teensyduino sketch to compile?

Am I using all the right library versions?

---------------------------------------
`collect2: error: ld returned 1 exit status
Using library Audio at version 1.3 in folder: /Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/Audio
Using library SPI at version 1.0 in folder: /Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/SPI
Using library Adafruit_GFX_Library at version 1.10.2 in folder: /Users/dwestbury/Documents/Tech_Stuff/Electronics/Arduino/libraries/Adafruit_GFX_Library
Using library Wire at version 1.0 in folder: /Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/Wire
Using library SD at version 1.2.2 in folder: /Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/SD
Using library MIDI at version 5.0.2 in folder: /Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/MIDI
Using library USBHost_t36 at version 0.1 in folder: /Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/USBHost_t36
Using library TeensyThreads at version 1.0.1 in folder: /Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/TeensyThreads
Using library CircularBuffer at version 1.3.3 in folder: /Users/dwestbury/Documents/Tech_Stuff/Electronics/Arduino/libraries/CircularBuffer
Using library Encoder at version 1.4.1 in folder: /Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/Encoder
Using library Bounce in folder: /Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/Bounce (legacy)
Using library ADC at version 8.0 in folder: /Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/ADC
Using library EEPROM at version 2.0 in folder: /Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/EEPROM
Using library SerialFlash at version 0.5 in folder: /Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/SerialFlash
Using library Adafruit_BusIO at version 1.6.0 in folder: /Users/dwestbury/Documents/Tech_Stuff/Electronics/Arduino/libraries/Adafruit_BusIO
Error compiling for board Teensy 4.1.
`

callimero, 2020-11-12:
Maybe not. :/ Which Arduino IDE are you using? Teensy audio has also some requirements, I installed 1.8.13 for TSynth.

Then check if you have the said libs installed (here in git in README).

For me it was I think a problem with the Circular buffer versions.

callimero, 2020-11-12:
Here is mine which compiles fine.
`
C:\Desktop_Ext\arduino-1.8.13\arduino-builder -dump-prefs -logger=machine -hardware C:\Desktop_Ext\arduino-1.8.13\hardware -hardware C:\Users\cw.Mero\AppData\Local\Arduino15\packages -hardware C:\Users\cw.Mero\Documents\Arduino\hardware -tools C:\Desktop_Ext\arduino-1.8.13\tools-builder -tools C:\Desktop_Ext\arduino-1.8.13\hardware\tools\avr -tools C:\Users\cw.Mero\AppData\Local\Arduino15\packages -built-in-libraries C:\Desktop_Ext\arduino-1.8.13\libraries -libraries C:\Users\cw.Mero\Documents\Arduino\libraries -fqbn=teensy:avr:teensy41:usb=serialmidiaudio,speed=600,opt=o2std,keys=en-us -ide-version=10813 -build-path C:\Users\CW7930~1.MER\AppData\Local\Temp\arduino_build_933994 -warnings=none -build-cache C:\Users\CW7930~1.MER\AppData\Local\Temp\arduino_cache_436801 -verbose C:\Users\cw.Mero\Documents\Makes\TSynth\TSynth-Teensy4.1\TSynth\TSynth.ino
C:\Desktop_Ext\arduino-1.8.13\arduino-builder -compile -logger=machine -hardware C:\Desktop_Ext\arduino-1.8.13\hardware -hardware C:\Users\cw.Mero\AppData\Local\Arduino15\packages -hardware C:\Users\cw.Mero\Documents\Arduino\hardware -tools C:\Desktop_Ext\arduino-1.8.13\tools-builder -tools C:\Desktop_Ext\arduino-1.8.13\hardware\tools\avr -tools C:\Users\cw.Mero\AppData\Local\Arduino15\packages -built-in-libraries C:\Desktop_Ext\arduino-1.8.13\libraries -libraries C:\Users\cw.Mero\Documents\Arduino\libraries -fqbn=teensy:avr:teensy41:usb=serialmidiaudio,speed=600,opt=o2std,keys=en-us -ide-version=10813 -build-path C:\Users\CW7930~1.MER\AppData\Local\Temp\arduino_build_933994 -warnings=none -build-cache C:\Users\CW7930~1.MER\AppData\Local\Temp\arduino_cache_436801 -verbose C:\Users\cw.Mero\Documents\Makes\TSynth\TSynth-Teensy4.1\TSynth\TSynth.ino
Using board 'teensy41' from platform in folder: C:\Desktop_Ext\arduino-1.8.13\hardware\teensy\avr
Using core 'teensy4' from platform in folder: C:\Desktop_Ext\arduino-1.8.13\hardware\teensy\avr
Detecting libraries used...
...
Opening Teensy Loader...
Mehrere Bibliotheken wurden für "SD.h" gefunden
 Benutzt: C:\Desktop_Ext\arduino-1.8.13\hardware\teensy\avr\libraries\SD
 Nicht benutzt: C:\Desktop_Ext\arduino-1.8.13\libraries\SD
Bibliothek Audio in Version 1.3 im Ordner: C:\Desktop_Ext\arduino-1.8.13\hardware\teensy\avr\libraries\Audio  wird verwendet
Bibliothek SPI in Version 1.0 im Ordner: C:\Desktop_Ext\arduino-1.8.13\hardware\teensy\avr\libraries\SPI  wird verwendet
Bibliothek Adafruit_GFX_Library in Version 1.6.1 im Ordner: C:\Users\cw.Mero\Documents\Arduino\libraries\Adafruit_GFX_Library  wird verwendet
Bibliothek Wire in Version 1.0 im Ordner: C:\Desktop_Ext\arduino-1.8.13\hardware\teensy\avr\libraries\Wire  wird verwendet
Bibliothek SD in Version 1.2.2 im Ordner: C:\Desktop_Ext\arduino-1.8.13\hardware\teensy\avr\libraries\SD  wird verwendet
Bibliothek MIDI in Version 5.0.2 im Ordner: C:\Desktop_Ext\arduino-1.8.13\hardware\teensy\avr\libraries\MIDI  wird verwendet
Bibliothek USBHost_t36 in Version 0.1 im Ordner: C:\Desktop_Ext\arduino-1.8.13\hardware\teensy\avr\libraries\USBHost_t36  wird verwendet
Bibliothek TeensyThreads in Version 1.0.1 im Ordner: C:\Desktop_Ext\arduino-1.8.13\hardware\teensy\avr\libraries\TeensyThreads  wird verwendet
Bibliothek CircularBuffer in Version 1.3.3 im Ordner: C:\Users\cw.Mero\Documents\Arduino\libraries\CircularBuffer  wird verwendet
Bibliothek Encoder in Version 1.4.1 im Ordner: C:\Desktop_Ext\arduino-1.8.13\hardware\teensy\avr\libraries\Encoder  wird verwendet
Bibliothek Bounce im Ordner: C:\Desktop_Ext\arduino-1.8.13\hardware\teensy\avr\libraries\Bounce (legacy) wird verwendet
Bibliothek ADC in Version 8.0 im Ordner: C:\Desktop_Ext\arduino-1.8.13\hardware\teensy\avr\libraries\ADC  wird verwendet
Bibliothek EEPROM in Version 2.0 im Ordner: C:\Desktop_Ext\arduino-1.8.13\hardware\teensy\avr\libraries\EEPROM  wird verwendet
Bibliothek SerialFlash in Version 0.5 im Ordner: C:\Desktop_Ext\arduino-1.8.13\hardware\teensy\avr\libraries\SerialFlash  wird verwendet
"C:\\Desktop_Ext\\arduino-1.8.13\\hardware\\teensy/../tools/arm/bin/arm-none-eabi-size" -A "C:\\Users\\CW7930~1.MER\\AppData\\Local\\Temp\\arduino_build_933994/TSynth.ino.elf"
Der Sketch verwendet 271216 Bytes (3%) des Programmspeicherplatzes. Das Maximum sind 8126464 Bytes.
Globale Variablen verwenden 279828 Bytes (53%) des dynamischen Speichers, 244460 Bytes für lokale Variablen verbleiben. Das Maximum sind 524288 Bytes.

`

dwestbury, 2020-11-12:
Thank you for the insight...

Looks like the only library difference is 'Adafruit_GFX_Library'
> I'm using 1.10.2 and you are able to compile successfully with 1.6.1

For completeness, I'm on a Mac with OSX Catalina 10.15.7
Arduino IDE: 1.8.13
Teensyduino: 1.53

I can try to downgrade the Ada GFX lib to see if that makes a difference.

dwestbury, 2020-11-13:
Apparently that was it... I installed 'Adafruit_GFX_Library' v1.6.1 and the code compiles properly.
Thanks for the insight @callimero, much appreciated.

closed 2020-11-13 by dwestbury