Electrotechnique/TSynth-Teensy4.1 #8: Release is "weird"

callimero, 2020-11-26:
Sorry for the fuzzy title.

The release seems to end a bit sudden especially when using relative short release times (0.2-1.0s). It seems then the release is maybe linear and that gives the ear the sudden end. Which does not sound natural.

I compared it with my other synths and there it seems more exponential leading to a smoother end.

When using a longer release (>2s or so)  it is better. Havn't looked at the code, maybe it is in the audio lib?

ElectroTechnique, 2020-12-18:
Yes, envelopes are linear. Exponential ones would require a re-write to the Audio Lib.
[https://forum.pjrc.com/threads/60428-Logarithmic-More-Natural-Fade-with-Release?highlight=exponential](Link to PJRC Forum)

callimero, 2020-12-18:
Ok, for now I slap a reverb behind it :)

closed 2020-12-18 by callimero