Electrotechnique/TSynth-Teensy4.1 #9: Display Image is Skewed on ST7735

dwestbury, 2020-11-29:
Has anyone experienced a skewed image on their LCD display?

[image]

The display doesn't appear to have any physical adjustments, so wondering if there's a way to fix this within the code?

Using library Audio at version 1.3...
Using library SPI at version 1.0...
Using library Adafruit_GFX_Library at version 1.6.1...
Using library Wire at version 1.0...
Using library SD at version 1.2.2...
Using library MIDI at version 5.0.2...
Using library USBHost_t36 at version 0.1...
Using library TeensyThreads at version 1.0.1...
Using library CircularBuffer at version 1.3.3...
Using library Encoder at version 1.4.1...
Using library Bounce...
Using library ADC at version 8.0...
Using library EEPROM at version 2.0...
Using library SerialFlash at version 0.5...
Using library Adafruit_BusIO at version 1.6.0...

-Darrell

sdh7, 2020-12-01:
I had the same issue. This may help-

In ST7735Display.h, in the setupDisplay() function at the bottom, change 
tft.initR(INITR_GREENTAB);

to 

tft.initR(INITR_BLACKTAB);

and then set tft.invertDisplay(false) instead of true, so that the background is black.

YMMV, as these cheapo displays apparently vary quite a bit. My display was a green-tab one, but had to be changed to INITR_BLACKTAB to work properly.

dwestbury, 2020-12-01:
Ah, brilliant guidance @sdh7 ... 

The recommended changes worked like a **_charm_**...

`void setupDisplay() {
  tft.useFrameBuffer(true);
  tft.initR(INITR_BLACKTAB);
  tft.setRotation(3);
  tft.invertDisplay(false);
  renderBootUpPage();
  tft.updateScreen();
  threads.addThread(displayThread);
}`

[image]

closed 2020-12-01 by dwestbury

Gerassic426, 2020-12-11:
> I had the same issue. This may help-
>
> In ST7735Display.h, in the setupDisplay() function at the bottom, change
> tft.initR(INITR_GREENTAB);
>
> to
>
> tft.initR(INITR_BLACKTAB);
>
> and then set [...]

THANK YOU SO MUCH MY GUY, this really helped me a lot, my screen had parts with random pixels and now it's smooth plus I have more space to work with!!! thank you, thank you, thank you!!