Electrotechnique/TSynth-Teensy4.1 #11: 100 Ohm Resistor on Display SCK is Not Eliminating Audio Noise?

dwestbury, 2020-12-13:
I was hearing the popping and static noise that others were complaining about, so I went ahead and cut the PCB trace for the display clock (SCK) and added in a 100 Ohm resistor as noise reduction bridge, but to my ears there's nearly no difference? 

[image]

Are there other possible sources of the noise? It's not terrible, but it is clearly perceptible.

ElectroTechnique, 2020-12-18:
Hello Darrell, are you sure that you've cut completely through the SCK track? I've compared before and after, and the resistor does make a significant reduction in noise.

[image]

dwestbury, 2020-12-18:
I actually peeled off the section of the copper trace leading from my cut on the board to the SDA pin.

Another area I need to investigate as a possible noise source is the dupont connector I made, so I can remove the teensy, rather than soldering it directly to the PCB. 

[image]

[image]

nfroeschl, 2020-12-29:
It looks like I have the same problem here. The resistor is in place but the noise (volume independent) is still there. 
It look like it soming from any kind of computer USB. With a powerbank away from the computer it is not there.

Any solutions so far ?

ElectroTechnique, 2020-12-31:
If it goes away when you change the power source, then it's coming from that. Be careful with ground loops.

ElectroTechnique, 2021-01-11:
I can confirm that the V1.3 PCB which has a slightly re-routed display SCK track with a 100ohm resistor, is much quieter. Bypassing the 100ohm resistor increases the clock noise significantly.

closed 2021-01-11 by ElectroTechnique