Electrotechnique/TSynth-Teensy4.1 #12: Aliexpress ST7735 Color display issues

SurvivalHacking, 2020-12-26:
If a display that appears to be compatible is used, but purchased from inexpensive vendors, such as Aliexpress, there could be malfunction issues. I was able to fix all the problems by making a small change to some definitions in a couple of files.

The first problem is a 1 pixel shift that bends the image. To fix this problem, I modified the function: **ST7735_t3::initR** in the file:  **ST7735_t3.cpp**

  if (options == INITR_GREENTAB) {
    commandList(Rcmd2green);
    _colstart = 2;
    _rowstart = **0**; // modified from 1 to 0 by Davide Gatti to fix Aliexpress display

A second problem is the color inversion, to fix this I modified two define in the file: **ST7735_t3.h**

#define ST7735_INVOFF **0x21** // Changed from 20 to 21 By Davide Gatti for Aliexpress display
#define ST7735_INVON **0x20** // Changed from 21 to 20 By Davide Gatti for Aliexpress Display

These 2 simple changes fixed the compatible display.

philippejadin, 2021-02-16:
would it be use full to provide this as a configurable option somehow ?

SurvivalHacking, 2021-02-16:
It seems superseeded by Electrotecnique
You need to change the following lines in ST7735Display.h

  tft.initR(INITR_BLACKTAB);  //INITR_GREENTAB 
  tft.invertDisplay(true);  // false
  
  I tested on V2.10 and works fine.

closed 2021-04-11 by ElectroTechnique