Electrotechnique/TSynth-Teensy4.1 #13: teensy 4.1 usb client wires pads

snowsh01, 2021-02-06:
Note to be extra careful when soldering the USB jumper wires to the Teensy 4.1 D+ and D- pads. Compared to previous Teensy boards, the 4.1 pads are tiny and very weak. any movement of the wires will likely result in the pads pulling off. 

Looking at the Teensy 4.1 schematics a there is fix in the event this occurs but I am yet to perform the fix. 

Fix:

If you have damaged the pads, you could connect D+ and D- via U5 pins 6 and 5 respectively

https://www.pjrc.com/teensy/schematic.html

[image]
[image]
[image]

philippejadin, 2021-02-16:
I destroyed mine quickly indeed. I'd advice to use very thin wire, solder them like shown in the user guide, and right after put a drop of hot glue to protect the pads and the wires. Too late for me :-/

snowsh01, 2021-02-16:
Yep. I killed mine and learnt the hard way.... Unfortunately there seems to be no easy way to solder in elsewhere

philippejadin, 2021-02-16:
It's not that bad, you can use the micro usb instead for conencting and programming, and if/when it get destroyed, you can unsolder it, and connect back the D+ D- wires from underneath the connector. I would only do it if the micro usb got destroyed because it's probably hard surgery to remove the micro usb connector without touching the rest. It's imho a negative point of teensy : they are too small :-)

antoniograzioli, 2021-02-25:
too late for me as well, i soldered them , they were working, but after some rremoval/reinserts of the teensy they fell apart...