Electrotechnique/TSynth-Teensy4.1 #18: What features are possible?

winder, 2021-03-20:
Sorry for the vague open-ended question, let me explain. I'm a programmer by day with some electronics and audio hobbies, recently I've been interested in playing with different sounds and stumbled upon the TSynth project. Looking at the code as a software developer I see a lot of opportunity for additional features (i.e. arpeggiator, mono mode + note priorities, etc).

If I spent the time on the code, what sort of features could the teensy handle it? An arpeggiator? What about a split mode?

In general, if you had all the time and inspiration in the world, what other features would be possible/reasonable?

I'm new to this hobby, so I was specifically looking at what some other polysynths have, like the Korg Minilogue:
[image]

ElectroTechnique, 2021-03-22:
I opted not to include an arpeggiator or sequencer because it's a MIDI synth and these features are better provided by other equipment usually. Multi-timbrality would be a good feature and shouldn't cause any extra CPU load or RAM issues.

remip, 2021-03-22:
A nice feature should be to add drum kits, with different sounds (presets) mapped to midi keys with multi-timbrality

winder, 2021-03-24:
Thanks for the feedback. Looking into the teensy a bit further, this board and the audio library are much more powerful than I realized.

I think a chord mode might be a good way to get started.

For multi timbral mode what exactly do you have in mind? Would I be able to configure a different patch per oscillator? I guess something like that would need to be configured on a computer, which isn't quite as fun as doing it all in the hardware.

callimero, 2021-03-24:
Yea, have fun! 

Keep in mind that for a "broad" user base you need to fit it in the Panel-UI. So basically encoder and three knobs. Too much menu diving will lessen the experience of the many knobs.

It took me some time to get how to extend the menus. (but I am no developers anyhow).

closed 2021-04-14 by ElectroTechnique