Electrotechnique/TSynth-Teensy4.1 #107: 11th Voice not working ??

nfroeschl, 2021-06-06:
Today I tried the Master-MT branch (9298d69) and I just played 1 note serveral tie and I realized that every 11th note was missing. I see on the voice grid that it is always the same voice (11th) which is mssing. Could someone try if they have the same problem with this build ?

ElectroTechnique, 2021-06-09:
Yes, it's been noticed. There are one or two bugs still. Unison & Monophonic causes crashes for instance.

ElectroTechnique, 2021-07-02:
Unison & Monophonic causing crashes is now fixed in V2.31. Haven't had the 11th (and 9th and 10th) voice missing problem in a while. A bit strange, it seemed to be caused by plugging into USB after being unplugged for a while?!?

glido1, 2021-07-04:
Contact tsynth with USB also causes DAW crashes, e.g. Cubase, and if it does not crash, the CPU load reaches 90%. I tested on win7 64 with cubase 5 and win 10 with cubase 10, After several hours of searching for the problem, I disconnected the USB and no more i use.
The Midi goes through the Motu interface and all TSYNTHs are plugged into the Motu I2S bus.
Two months have passed and I have no problems.
Additionally, I can use the 32 bit DSP database offered by MOTU.

closed 2021-07-26 by ElectroTechnique