Electrotechnique/TSynth-Teensy4.1 #121: Unable to Compile Firmware v2.33

dwestbury, 2022-08-21:
I have the BLACKTAB display issue, so whenever a new firmware becomes available, I need to edit the ST7735Display.h file and recompile.

Running into the following issue when compiling for v2.33...

**From the Teensyduino IDE:**

`Arduino: 1.8.13 (Mac OS X), TD: 1.53, Board: "Teensy 4.1, Serial + MIDI + Audio, 600 MHz, Faster, US English"

/var/folders/d5/5n5zgcmd6svfhnzwcqx2jjdc0000gn/T/arduino_build_91510/sketch/synth_waveform.cpp.o: In function `BandLimitedWaveformTS::lookup(int)':
**/var/folders/d5/5n5zgcmd6svfhnzwcqx2jjdc0000gn/T/arduino_build_91510/sketch/synth_waveform.cpp:526: undefined reference to `step_table'**
collect2: error: ld returned 1 exit status

Using library Audio at version 1.3 in folder: /Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/Audio 
Using library SPI at version 1.0 in folder: /Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/SPI 
Using library Adafruit_GFX_Library at version 1.11.3 in folder: /Users/dwestbury/Documents/Tech_Stuff/Electronics/Arduino/libraries/Adafruit_GFX_Library 
Using library Adafruit_BusIO at version 1.6.0 in folder: /Users/dwestbury/Documents/Tech_Stuff/Electronics/Arduino/libraries/Adafruit_BusIO 
Using library Wire at version 1.0 in folder: /Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/Wire 
Using library SD at version 1.2.2 in folder: /Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/SD 
Using library MIDI at version 5.0.2 in folder: /Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/MIDI 
Using library USBHost_t36 at version 0.1 in folder: /Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/USBHost_t36 
Using library TeensyThreads at version 1.0.1 in folder: /Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/TeensyThreads 
Using library CircularBuffer at version 1.3.3 in folder: /Users/dwestbury/Documents/Tech_Stuff/Electronics/Arduino/libraries/CircularBuffer 
Using library Encoder at version 1.4.1 in folder: /Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/Encoder 
Using library Bounce in folder: /Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/Bounce (legacy)
Using library ADC at version 8.0 in folder: /Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/ADC 
Using library EEPROM at version 2.0 in folder: /Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/EEPROM 
Using library SerialFlash at version 0.5 in folder: /Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/SerialFlash 
Error compiling for board Teensy 4.1.
`

ElectroTechnique, 2022-08-24:
Hello, can you update to the latest version of Arduino IDE and Teensyduino and compile?

dwestbury, 2022-08-25:
Updating my Arduino IDE to 1.8.19 and Teensyduno to 1.57 seems to have done the trick...

[image]

I'm able to compile successfully now...

[image]

Cheers!

closed 2022-08-25 by dwestbury