Electrotechnique/TSynth-Teensy4.1 #126: Feature Request: ability to select screen "type" without uploading different versions of firmware

worsco, 2023-04-23:
I was thinking about a method of configuring the display during initial boot instead of having to upload different versions of the firmware.

There are three buttons below the display: "Settings", "Save", and "Back" and to the right of the display is the rotary encoder which has the ability to be pressed to "confirm" a selection.

Binary math provides us with eight (8) different "settings" on power-up:

```
   3 2 1*
0: 0 0 0 no buttons pressed power-up == no state change
1: 0 0 1 default
2: 0 1 0
3: 0 1 1
4: 1 0 0
5: 0 1 0
6: 1 1 0
7: 1 1 1

*Buttons:
1 == Settings
2 == Save
3 == Back
```

On initial boot-up, what if the three buttons were used to select one of seven possible screen settings (not that there are 7 current display setting possibilities, but there is the ability to select that many different config options on initial power-up).  

Here is my pseudo code:
```
1. If any of the three buttons are pressed on power-up: 
   a. based on the button pressed, choose the configuration setting and configure display
   b. initialize screen with setting chosen
   c. on the display, ask confirmation question "Is display correct/viewable?"
      1. 10 second loop count down
      2. during loop, press down on rotary to confirm
      3. If rotary pressed down, 
         a. if SD card exists, then
            1. create file screen setting (or write-over existing file)
            2. write screen configuration setting (assumes one file used the screen setting)
            3. close file
         b. else exit loop & continue
      4. after 10 second loop (rotary never pressed), continue  w/o saving
         (perhaps the logic could it loop into a sequence where you can try another key combo,
          or would it be better to revert to the "default" screen config and continue booting?)

3. if no buttons pressed on initial power-up
   a. look for config file on SD card for screen configuration file
   b. if file found
       1. parse contents
       2. validate setting:
          a. if valid, set screen config to setting chosen
          b. if invalid, use "default" screen setting (from a previously set constant variable)
```

worsco, 2023-04-23:
Maybe the first iteration of configuring the screen would to just implement the "Settings/Save/Back" button combos to select which configuration to use on power-on.  The next phase would to make the setting persistent after selecting.

wilksy, 2023-04-23:
I've actually done this. Well, I've actually rewritten most of the display code to work with animated dials and visual graphs for the waveforms.

Give me a week it two and I'll figure out how to upload, I've actually never hues GitHub for Collab projects.

Unless someone fancies taking the code and working with it?

worsco, 2023-04-23:
> Give me a week it two and I'll figure out how to upload, I've actually never hues GitHub for Collab projects.
> 
> Unless someone fancies taking the code and working with it?

@wilksy 

Did you fork this repo in github?  If so, you can post your changes to your fork and the submit PRs (Pull Requests) to the original repo.  If you post your code in your own repo, I can take a look and see your handy work.

Thanks in advance.

Electrotechnique, 2026-11-14:
Fixed in forthcoming V3 firmware. Any other display problems, please report to Electrotechnique - info@Electrotechnique.cc

closed 2026-11-14 by Electrotechnique