Electrotechnique/TSynth-Teensy4.1 #130: error: ordered comparison of pointer with integer zero ('char* (*)(const char*, int)' and 'int')

danielkinahan, 2023-10-06:
Trying to verify this project and getting this error among others. Some info

- Arduino IDE 2.2.1
- Teensyduino 1.58.1
- Using latest main commit e2fd7cd
- Arduino IDE Tools Settings:
  - Board: "Teensy4.1"
  - USB Type: "Serial + MIDI + Audio"
  - CPU Speed: "600MHz"
  - Optimize: "Faster"
- CircularBuffer - AgileWare: 1.3.3
- Adafruit GFX Library: 1.11.8
- Windows 11

Full error message:
```
C:\TSynth-Teensy4.1\TSynth\effect_ensemble.cpp: In member function 'int16_t AudioEffectEnsemble::interpBuffer(float)':
C:\TSynth-Teensy4.1\TSynth\effect_ensemble.cpp:267:16: error: ordered comparison of pointer with integer zero ('char* (*)(const char*, int)' and 'int')
  267 |   else if(index<0) index2+=ENSEMBLE_BUFFER_SIZE;
      |           ~~~~~^~
In file included from C:\TSynth-Teensy4.1\TSynth\Audio.h:67,
                 from C:\TSynth-Teensy4.1\TSynth\TSynth.ino:48:
C:\TSynth-Teensy4.1\TSynth\synth_dc.h: In member function 'void AudioSynthWaveformDcTS::amplitude(float, float)':
C:\TSynth-Teensy4.1\TSynth\synth_dc.h:118:17: error: 'void AudioSynthWaveformDcTS::amplitude(float, float)' causes a section type conflict with 'int compare(const void*, const void*)'
  118 |   FLASHMEM void amplitude(float n, float milliseconds) {
      |                 ^~~~~~~~~
In file included from C:\TSynth-Teensy4.1\TSynth\TSynth.ino:60:
C:\TSynth-Teensy4.1\TSynth\PatchMgr.h:74:14: note: 'int compare(const void*, const void*)' was declared here
   74 | FLASHMEM int compare(const void *a, const void *b) {
      |              ^~~~~~~
```

danielkinahan, 2023-10-07:
I have verified that I'm also getting this on the latest released tag V2.33

danielkinahan, 2023-10-07:
For the first error it's now obvious to me after looking at the code that `index` should be `index2`. I will make a PR with that simple change soon.

I am leaning towards thinking the second error is the result of my build steps. In the FLASHMEM section 
`PatchMgr.h - int compare(const void*, const void*)` 
is already declared and causing a section type conflict with
`synth_dc.h - void AudioSynthWaveformDcTS::amplitude(float, float)`

Research for this tells me to declare these objects into different sections initially and combine them with a linker instruction but I'm not sure how to achieve that in this codebase. I suspect that something in TeensyDuino might have changed since the FLASHMEM definition comes from that library.

ElectroTechnique, 2023-11-20:
Fixed in V2.35.

closed 2023-11-20 by ElectroTechnique